au-ts/sddf #53: Improvements to serial sub-system

Ivan-Velickovic, 2024-03-14:
- [ ] no busy waiting in serial server example
- [ ] get rid of cookie in shared ring buffer.
- [ ] rename ring_* to serial_ring_*
- [ ] Document RAW_MODE, ECHO_MODE, LINE_MODE etc
- [ ] We really need some systematic logging
- [ ] Try not comment out assert and puthex64
- [ ] Use `sddf_dprintf` instead of `microkit_dbg_puts` for easier printing
- [ ] Fixup serial example Makefile to call into queue Makefile
- [ ] All the state stuff needs to be cleaned up in the drivers and virtualisers, it is hard to decipher what is in what state when.
- [ ] virtualisers dequeue only one used entry at a time
- [ ] use offsets instead of raw addresses

Courtney3141, 2024-06-13:
Add a debug print statement on the rx virt when a switch char has been received.

Ivan-Velickovic, 2024-06-13:
Done in https://github.com/au-ts/sddf/pull/99.

closed 2024-06-13 by Ivan-Velickovic