au-ts/sddf #113: VirtIO network driver peformance.

Kswin01, 2024-05-14:
Whilst testing the VirtIO net driver with the UDP echo socket, we ended up stalling the driver once requested through-puts reached approximately 600-700 Mbps and with 100,00 samples. It is able to match the requested through-put up until this limit. On inspection we stopped receiving IRQ's from QEMU before we completed the ipbench run. 

On testing with the TCP echo server, we don't stall the driver when tested up to 1 Gbps and with 200,000 samples.

Ivan-Velickovic, 2024-07-02:
Noticed by, @alwin-joshy, there is an overflow in the ARM timer driver here: https://github.com/au-ts/sddf/blob/52bcc48316b12705a01ace8c0045e20f98558739/drivers/clock/arm/timer.c#L95.

This may be why we could never finish an echo server run. Once we fix the bug we should re-run and see if anything has changed with the echo server.

Ivan-Velickovic, 2024-10-09:
With https://github.com/au-ts/sddf/pull/266, the UDP numbers are:
```
100000000,99994285,99999285,1472,86,494,3460,228.74,469 ,0,0
200000000,199984546,199998544,1472,333,641,5116,289.10,598 ,0,0
300000000,299946307,299997298,1472,306,643,6281,316.94,593 ,0,0
400000000,399883729,399999695,1472,360,726,2909,258.85,680 ,0,0
500000000,499748131,499998005,1472,378,841,4086,319.11,827 ,0,0
600000000,599667168,599996985,1472,391,861,2883,299.64,840 ,0,0
700000000,699658013,699993849,1472,388,1352,7841,712.73,1232 ,0,0
800000000,799742474,799998391,1472,398,1636,7882,856.23,1471 ,0,0
900000000,896201645,899992578,1472,509,3952,9586,1290.61,3904 ,0,0
1000000000,992443741,999996237,1472,2725,12128,37066,6147.70,10852 ,0,0
```

and the TCP numbers are:
```
100000000,99996501,99999501,1472,127,510,7637,207.81,532 ,0,0
200000000,199995822,199999822,1472,172,547,2895,176.10,537 ,0,0
300000000,299969878,299999875,1472,237,636,8097,366.90,607 ,0,0
400000000,399958428,399998424,1472,294,858,7730,713.75,742 ,0,0
500000000,499837964,499997912,1472,747,1762,6510,908.70,1485 ,0,0
600000000,388616724,397041935,1472,1235,122060,1546465,209040.68,97280 ,0,0
700000000,508475710,526341004,1472,1142,151963,169546,29879.47,160668 ,0,0
800000000,507738856,519183290,1472,1268,101703,113283,14023.54,104440 ,0,0
900000000,504041081,515936451,1472,1265,105637,120383,13518.83,107981 ,0,0
1000000000,508010461,520256053,1472,1430,109043,123426,13338.07,111302 ,0,0
```

Not sure why TCP is only at half a gigabit.

Ivan-Velickovic, 2025-05-19:
With #429:

UDP:
```
Requested_Throughput,Receive_Throughput,Send_Throughput,Packet_Size,Minimum_RTT,Average_RTT,Maximum_RTT,Stdev_RTT,Median_RTT,CPU_utilisation
100000000,99999656,99999656,1472,63,238,1434,131.73,225,0 ,0,0
200000000,200000560,199999560,1472,85,572,1904,142.10,560,0 ,0,0
300000000,299995248,299999748,1472,139,568,2623,148.63,556,0 ,0,0
400000000,400022147,400000146,1472,155,612,2596,192.25,588,0 ,0,0
500000000,500005201,500000201,1472,376,823,4769,293.19,807,0 ,0,0
600000000,600011515,599999514,1472,169,1106,4939,507.85,1037,0 ,0,0
700000000,700505808,700001444,1472,490,3445,11089,2177.56,2764,0 ,0,0
800000000,799741659,800001575,1472,460,3977,9437,1363.45,3903,0 ,0,0
900000000,900309940,899999333,1472,4039,10860,38684,2963.24,10667,0 ,0,0
1000000000,884567209,999998807,1472,136956,144946,152391,2831.32,144953,0 ,0,0
```

TCP:
```
Requested_Throughput,Receive_Throughput,Send_Throughput,Packet_Size,Minimum_RTT,Average_RTT,Maximum_RTT,Stdev_RTT,Median_RTT,CPU_utilisation
100000000,100001931,99999931,1472,92,351,2671,152.55,357,0 ,0,0
200000000,199995652,199999652,1472,120,424,1636,162.71,438,0 ,0,0
300000000,299988026,300000026,1472,147,467,2154,152.01,479,0 ,0,0
400000000,399999184,400001184,1472,175,629,7740,539.38,564,0 ,0,0
500000000,499950221,500000216,1472,222,891,8450,1038.80,618,0 ,0,0
600000000,599581830,599998539,1472,6025,33062,53672,13276.42,35054,0 ,0,0
700000000,630052700,630216514,1472,82825,88264,107506,3627.86,87542,0 ,0,0
800000000,621722704,621747572,1472,105305,112194,130331,5025.38,110672,0 ,0,0
900000000,598198293,598488419,1472,83846,94054,110337,5211.61,93444,0 ,0,0
1000000000,630173573,630050689,1472,90065,96484,106203,2677.45,96230,0 ,0,0
```