au-ts/sddf #130: net_queue_size can return a wrong value

alainkaegi, 2024-06-12:
`net_queue_size` can return the wrong value when the `tail` index has wrapped around, but `head` has not.

The function must take the queue size into account (i.e., `(tail - head) % size`).

Courtney3141, 2024-06-19:
Actually I think this is due to the function returning a uint32 rather than uint16 which is the type of the head and tail

alainkaegi, 2024-06-22:
Yes, I agree that changing the return type of `net_queue_size` to uint16 appears to work.

BTW using "size" to mean slightly different things depending on the context makes the code unnecessarily harder to read.  In one context (struct definition), it is related to the maximum number of entries that a queue can hold; in the other (function name), it indicates the number of entries currently held in the queue.  Therefore I would suggest changing the field name in `net_queue_handle_t` from `size` to `capacity` (and change the comment accordingly as well as the argument to `net_queue_init`).

Ivan-Velickovic, 2024-07-02:
@alainkaegi is right. We need to start distinguishing between queue size and capacity. I've made https://github.com/au-ts/sddf/issues/140.