au-ts/sddf #138: I2C sub-system

Ivan-Velickovic, 2024-06-27:
### I2C sub-system

* [x] REQ_BUF_CLIENT and REQ_BUF_ADDR should be part of the shared ring buffer protocol instead as metadata.
* [x] Too many files exist, it should really just be driver.c and server.c
* [x] Clean up example system description file - lots of unused memory regions, needs comments as well
* [ ] Getting timeout IRQs way too frequently - these should never happen if everything is going smoothly
  to my understanding.
* [x] The I2C server needs to be fully designed and implemented. Right now the client talks directly to the driver.
* [ ] Remove all `@ivanv` notes.
~~* [ ] Currently the I2C driver is responsible for setting up the clocks, etc. This is obviously wrong and should be
  done in pinctrl/regulator/clock drivers. However, this is not something we have right now.~~
* [ ] Should we abstract over doing things like writing to an I2C bus (e.g setting up all the transaction tokens)
* [x] We should write a little client side API for claiming and release a bus.
~~sDDF I2C example should use queue Makefile~~ no more `queue.c`, no more Makefile!

### Example client notes
* [x] Timer driver integrated so we're not busy-waiting
* [x] Figure out that extra response dequeue when we read_response
* [ ] What to do about just copying libco into the source
* [ ] More error-checking and cleanup is needed
* [x] We are running out of return buffers fairly quickly, we must have a leak somewhere.
* [x] Remove all warnings when compiling the example.

### Driver notes
* [x] Get rid of popBufRet abstractions, etc
* [ ] Too much global state and it's too fragile

### Documentation
* [ ] The documentation is a great start but is a bit too verbose and needs to be consistent with the implementation.
* [ ] Need picture of card-reader working and exactly what card reader is used (with links)
* [x] Need to document that the card-reader client code is based of the Arduino PN532 library.

TristanCM347, 2024-07-04:
Card reader is the one giving the timeout interupts not the ds3231. There also seems to be a consistent pattern to them.

TristanCM347, 2024-07-04:
the zig make file needs updating