au-ts/sddf #295: Figure out what to do with the ARP component

Ivan-Velickovic, 2024-12-12:
Earlier this year we removed the ARP component from the echo server example since it was not necessary since lwIP can handle all the ARP packets.

The component code is now bit-rotted and so we need to figure out whether we want to keep supporting it by at least having a test for it or a separate example, or remove it altogether.

Willmish, 2025-09-05:
This got raised again while I was cleaning up components to use OS-agnostic wrappers in the network class (https://github.com/au-ts/sddf/pull/491). 

Is there any reason to keep the arp component files?

https://github.com/au-ts/sddf/blob/main/network/components/arp.c
https://github.com/au-ts/sddf/blob/main/include/sddf/network/arp.h

Ivan-Velickovic, 2025-09-05:
At the time of making this issue I think the reason we didn't remove it was because different IP stacks may not handle ARP themselves.

Courtney3141, 2025-09-05:
It's basically a choice as to whether you assume that clients handle broadcast traffic themselves (and expect it), or whether there is a system wide broadcast traffic handler. Given that we currently multiplex based on MAC addresses, it seems that clients would expect they receive all traffic, including broadcast.

Courtney3141, 2025-09-08:
Although it does add complexity, it may not be a bad idea for the virtualiser to be aware of the locations of buffers throughout the system. 

Re new attack vectors, I don't believe this is true, since clients are always required to have a copy component which stops clients from being able to hoard buffers (hoarding buffers is always a problem if a copy component is not present, regardless of how we handle broadcast).