au-ts/sddf #343: set_baud in imx serial driver causes garbled output on maaxboard

adam-yt, 2025-01-30:
I was testing how the imx serial driver runs in `examples/serial` on a maaxboard, but the output after being dropped into user space was garbled:

[image]

After fiddling around in `drivers/serial/imx/uart.c` it seems to be mistake in `set_baud` since removing the line `set_baud(config.default_baud)` in `uart_setup` fixed the issue. 

This was in a minicom terminal and the baud rate was set to 115200 which is also the value of `config.default_baud`.

Another issue is an undefined macro `MICROKIT_PD_NAME_LENGTH` in both `include/sddf/benchmark/config.h` and  `include/sddf/serial/config.h` which I just set to 62 for the purpose of getting the example to build.

adam-yt, 2025-01-30:
Forgot to mention: this was on `main`.

Ivan-Velickovic, 2025-01-31:
> MICROKIT_PD_NAME_LENGTH

It is defined in the version of Microkit that sDDF expects, see the README for details. We have to temporarily pin to a commit of Microkit rather than a released version until the next release (which should happen within the next week or two).

Can you try with that version of the SDK and then report if you get the same issue? We also run regularly on MaaXBoard and see no issues with output.

adam-yt, 2025-01-31:
I cloned the repo again and tried with the Microkit version in the dependencies and the issue persists.

Ivan-Velickovic, 2025-02-03:
Right. Okay we'll try debug what's going on. I'm not sure what could be happening given that the minicom baud rate is the same as what the driver is setting it to.

Ivan-Velickovic, 2025-02-05:
I've just tried with a MaaXBoard locally now. I used the following minicom command: `minicom -D /dev/tty.usbserial-A50285BI -b 115200` and didn't see jumbled output. The main problem is that no colour is shown (probably an issue with minicom) but all the text is there at least.

I used the U-Boot that comes with the Linux image from [here](https://www.hackster.io/monica/getting-started-with-maaxboard-ca362d#toc-2--flash-the-sd-card-1), version 2018.03 according to the boot logs.

If you're using a USB to UART converter, is the right voltage (3.3V I believe)? What minicom command did you use to connect?

adam-yt, 2025-02-05:
I'm using a USB to UART converter with 5V and I used the `minicom -D /dev/ttyUSB1` to connect. The baud is set to 115200 by default as shown in the settings. How does the voltage affect the output?

Ivan-Velickovic, 2025-03-19:
While I can't reproduce the issue on the MaaXBoard, I've just seen the issue appear on the i.MX8MQ-EVK. This is good, as it means there is definitely something wrong with `set_baud`. Likely one of the defines is different per-board rather than the same for all i.MX8-based SoCs.

Ivan-Velickovic, 2025-04-14:
Hi @adam-yt can you try using `minicom -D /dev/ttyUSB1 -b 115200` and using 3.3V instead of 5V? If you have a UART converter like [1] you can move the jumper to make it use 3.3V instead of 5V.

When we have been able to see jumbled output, it looks like this:
```
MON|INFO: completed system invocations
Begin input
 input
[+ˋ
   R.,ZY[0
'client1' is client 1
Hello world! I am client1.
Please give me character!
Hello world! I am client0.
Please give me character!
```

where it fixes itself after initialisation. We're looking into the root cause of this but it seems like in your case the output is *always* jumbled?

[1]: https://external-content.duckduckgo.com/iu/?u=https%3A%2F%2Fbotland.com.pl%2F71085-large_default%2Fconverter-usb-uart-ftdi-ft232rl-usb-plug-waveshare-6170.jpg&f=1&nofb=1&ipt=aa4ff830484db2c92cfec8863142a481ba59a5365e1d9d9e7909ce13f3ded249

adam-yt, 2025-04-30:
I tried running it with a 3.3V USB to UART converter and no longer got garbled output. When using the 5V converter the output after initialization is always garbage and I have to reboot the board to get normal output again.

Ivan-Velickovic, 2025-05-06:
Thanks for the reply.

I believe that is the source of the issue, the voltage must match what the board supplies for UART which is 3.3V on the MaaXBoard.

I'll close this then. Feel free to open more issues if you run into anything else.

closed 2025-05-06 by Ivan-Velickovic